rapyuta-robotics/turtlebot3-UE #20: Using different Vehicle in CARLA project

thejeshk, 2022-01-14:
Hi, 

I find the integration with ROS2 and UE4&5 would be great. Thanks for the rclUE plug in. I am planning to test it to use ROS2. But I have few questions: 
1. What are the things to keep to use different vehicle that turtlebot3. For example: construction equipment (loader for example) 
2. What is the difficulty level when we try to use CARLA along with rclUE ? or Airsim ?

It would be really great if you could answer these questions?

Thank you :)

yuokamoto, 2023-04-28:
Sorry for not responding long time.
- I'm planning to add tutorials to create custom robot and other equipment. Since it is base on UE, you can find non robot asset such as conveyor in UE market place
- I haven't investigate deeply on CARLA and Airsim. It maybe not very simple. Putting turtlebot3 and other ROS2 enabled robot is possible but may have some limitation since some features requries GameMode and etc.

yuokamoto, 2023-08-12:
closed with no activity

closed 2023-08-12 by yuokamoto