rapyuta-robotics/turtlebot3-UE #55: Unreal editor crash

james-yoo, 2023-03-01:
Hi,
I am testing this repo under following condition:
```
- Ubuntu 20.04.5
- Unreal Engine 5.1.0
- Nvidia graphic driver 525.78.01
- ROS 2 Foxy (apt install ros-foxy-desktop)
- ROS 2 Navigation2 (apt install ros-foxy-navigation2)
- ROS 2 Turtlebot3 (apt install ros-foxy-turtlebot3)
```
I followed setup guide describe [here](https://github.com/rapyuta-robotics/turtlebot3-UE#setup-and-run) and packages are prepared as below:
```
- turtlebot3-UE(branch: devel)
- Plugin : rclUe(branch: devel)
- Plugin : RapyutaSimulationPlugins(branch: devel)
```
When press the 'Play' button to start simulation, the Unreal Editor is crashed as follow:
[image]

I attach the entire log
[crash_log.txt](https://github.com/rapyuta-robotics/turtlebot3-UE/files/10857384/crash_log.txt)

Any suggestions would be greatly appreciated.

SaundersJE97, 2023-04-27:
I have the same issue

yuokamoto, 2023-04-29:
Did you source /opt/ros/foxy/setup.bash before starting editor?

yuokamoto, 2023-08-12:
close since no activity

closed 2023-08-12 by yuokamoto

pabsan-0, 2024-06-06:
I've had this issue too. The solution was not to source ros in my `.bashrc`