rapyuta-robotics/turtlebot3-UE #57: Build and run:  fast-discovery-server tool not found!

GreenKkaa, 2023-03-16:
Hi, I am following tutorial and on the step "Build and Run" I have error message:

Build and run (after I fill UE and exe paths):
```
cd turtlebot3-UE
export UE5_DIR=<path to UE5>
./update_project_files.sh
./make_editor.sh
./run_editor.sh $(pwd) <ue_exe>
```

my first lines of output:

```
Server creation failed with the given settings. Please review locators setup.2023-03-16 16:11:40.578 [
PARTICIPANT Error] Problem creating RTPSParticipant -> Function createParticipant
fastdds discovery <discovery-args>

1
 fast-discovery-server tool not found!
```

It is suspicious because after - UE is opening uproject, all compiles and even runs. But does not send any ros2 messages I can see using `ros2 topic echo /scan`. But /scan topics is seen in `ros2 topic list` and I see some scan visualization in game view.

please help me - I actually looking for ability to create functionality with the same idea - UE app listen some msg, visualize something and post another messages. But after running example - it is silent. Thanks =)

yuokamoto, 2023-04-28:
Sorry for late replay.
Are you working on Ubuntu 20.04 or Ubuntu 22.04?

yuokamoto, 2023-08-12:
closed since no activity

closed 2023-08-12 by yuokamoto