rapyuta-robotics/turtlebot3-UE #59: Export Blueprint Turtlebot3

Juliette1012, 2023-04-30:
Hello, 

I would like to be able to simulate a turtlebot3 in Carla Simulator using Carla-ROS-Bridge. 
For this I need the blueprint and the functionality of the turtlebot in Unreal Engine. I saw your project and you managed to simulate a turtlebot and its different functionalities in Unreal engine. 
Do you know how I can use your project to do the same thing in the Carla simulator?

Thank you in advance for your help

Tadinu, 2023-05-01:
You seem to have referred to an UE practice that allows us to commonly use BPs/assets under Contents among projects, if so it might be better asked in UE forum. 
Some suggestions may be:
* Creating a BP of your own similar to TB3 in a Carla's project, ideally in a Plugin to be usable by other modules.
* A more straightforward way but might not be fully supported is In your Carla project's Contents, create symlinks to subfolders of turtlebot3_UE's Contents so you could access those TB3 BPs.

yuokamoto, 2023-05-01:
FYI: one pr is in progress https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/pull/154

I confirmed turtlebot3 can be contrlled from /cmd_vel topic in UE5 Empty level with GameModeBase.

yuokamoto, 2023-08-12:
closed since no activity

closed 2023-08-12 by yuokamoto