rapyuta-robotics/turtlebot3-UE #64: ROS2 Rviz Error : Global Status Warn Fixed Frame No tf data. Actual error: Frame [world] does not exist

YuminosukeSato, 2023-06-07:
[ROS2 Rviz Error : Global Status Warn Fixed Frame No tf data. Actual error: Frame [world] does not exist]

I got this error in Rviz.

How I solve this error?

```
./run_editor.sh <false or true to use dds server or not> $(pwd) <ue_exe>
cd turtlebot3-UE && source fastdds_setup.sh
ros2 launch nav2_bringup tb3_simulation_launch.py use_simulator:=False map:=<path to turtlebot3-UE>/Content/Turtlebot3_benchmark.yaml 
```
I did this command.
OS version :ubuntu 20.04
unreal engine5.1.1

yuokamoto, 2023-06-08:
Do you see turtlebot3 and lidar rays(red dots) in the scene?
[image]

yuokamoto, 2023-06-08:
Can you share the screenshot when you `Play` the simulation?

YuminosukeSato, 2023-06-10:
[image]
[image]

YuminosukeSato, 2023-06-10:
I got a error when  I Play the simulation.
`LoginId:d69c7d530464480cb426a9ddc047c2a3-000003e8
EpicAccountId:

Caught signal 11 Segmentation fault

libUnrealEditor-Core.so!std::__1::error_category::equivalent(int, std::__1::error_condition const&) const(+0x7)
libfastrtps.so.2!_ZNSt12system_errorC1ESt10error_codePKc(+0x41)
libfastrtps.so.2!UnknownFunction(0x3b926c)
libfastrtps.so.2!_ZN8eprosima7fastdds4rtps14UDPv4Transport22OpenAndBindInputSocketERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEtb(+0x932)
libfastrtps.so.2!_ZN8eprosima7fastdds4rtps21UDPTransportInterface26CreateInputChannelResourceERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_8fastrtps4rtps9Locator_tEbjPNS1_26TransportReceiverInterfaceE(+0x65)
libfastrtps.so.2!_ZN8eprosima7fastdds4rtps21UDPTransportInterface23OpenAndBindInputSocketsERKNS_8fastrtps4rtps9Locator_tEPNS1_26TransportReceiverInterfaceEbj(+0x122)
libfastrtps.so.2!_ZN8eprosima7fastdds4rtps14UDPv4Transport16OpenInputChannelERKNS_8fastrtps4rtps9Locator_tEPNS1_26TransportReceiverInterfaceEj(+0xfd)
libfastrtps.so.2!_ZN8eprosima8fastrtps4rtps16ReceiverResourceC2ERNS_7fastdds4rtps18TransportInterfaceERKNS1_9Locator_tEj(+0x79)
libfastrtps.so.2!_ZN8eprosima8fastrtps4rtps14NetworkFactory22BuildReceiverResourcesERNS1_9Locator_tERSt6vectorISt10shared_ptrINS1_16ReceiverResourceEESaIS8_EEj(+0x87)
libfastrtps.so.2!_ZN8eprosima8fastrtps4rtps19RTPSParticipantImpl23createReceiverResourcesERNS1_13LocatorList_tEbb(+0xb7)
libfastrtps.so.2!_ZN8eprosima8fastrtps4rtps19RTPSParticipantImplC1EjRKNS1_25RTPSParticipantAttributesERKNS1_12GuidPrefix_tES8_PNS1_15RTPSParticipantEPNS1_23RTPSParticipantListenerE(+0xc64)
libfastrtps.so.2!_ZN8eprosima8fastrtps4rtps19RTPSParticipantImplC1EjRKNS1_25RTPSParticipantAttributesERKNS1_12GuidPrefix_tEPNS1_15RTPSParticipantEPNS1_23RTPSParticipantListenerE(+0x18)
libfastrtps.so.2!_ZN8eprosima8fastrtps4rtps10RTPSDomain17createParticipantEjbRKNS1_25RTPSParticipantAttributesEPNS1_23RTPSParticipantListenerE(+0x105b)
libfastrtps.so.2!_ZN8eprosima8fastrtps6Domain17createParticipantERKNS0_21ParticipantAttributesEPNS0_19ParticipantListenerE(+0x13f)
librmw_fastrtps_shared_cpp.so!UnknownFunction(0x802c5)
librmw_fastrtps_shared_cpp.so!_ZN23rmw_fastrtps_shared_cpp18create_participantEPKcmPK22rmw_security_options_tbS1_PN14rmw_dds_common7ContextE(+0x5c2)
librmw_fastrtps_cpp.so!UnknownFunction(0x17edc)
librmw_fastrtps_cpp.so!UnknownFunction(0x1869f)
librmw_fastrtps_cpp.so!rmw_create_node(+0x11b)
librmw_implementation.so!rmw_create_node(+0x88)
librcl.so!rcl_node_init(+0xb61)
librclc.so!rclc_node_init_with_options(+0x1a9)
librclc.so!rclc_node_init_default(+0x171)
libUnrealEditor-rclUE.so!UROS2NodeComponent::Init() [/home/iplslam/turtlebot3-UE/Plugins/rclUE/Source/rclUE/Private/ROS2NodeComponent.cpp:110]
libUnrealEditor-RapyutaSimulationPlugins.so!URRRobotROS2Interface::InitRobotROS2Node(ARRBaseRobot*) [/home/iplslam/turtlebot3-UE/Plugins/RapyutaSimulationPlugins/Source/RapyutaSimulationPlugins/Private/Robots/RRRobotROS2Interface.cpp:127]
libUnrealEditor-RapyutaSimulationPlugins.so!URRRobotROS2Interface::Initialize(ARRBaseRobot*) [/home/iplslam/turtlebot3-UE/Plugins/RapyutaSimulationPlugins/Source/RapyutaSimulationPlugins/Private/Robots/RRRobotROS2Interface.cpp:38]
libUnrealEditor-RapyutaSimulationPlugins.so!ARRBaseRobot::InitROS2InterfaceImpl() [/home/iplslam/turtlebot3-UE/Plugins/RapyutaSimulationPlugins/Source/RapyutaSimulationPlugins/Private/Robots/RRBaseRobot.cpp:250]
libUnrealEditor-RapyutaSimulationPlugins.so!ARRBaseRobot::InitROS2Interface() [/home/iplslam/turtlebot3-UE/Plugins/RapyutaSimulationPlugins/Source/RapyutaSimulationPlugins/Private/Robots/RRBaseRobot.cpp:236]
libUnrealEditor-Engine.so!AController::Possess(APawn*) [/home/iplslam/UnrealEngine/Engine/Source/./Runtime/Engine/Private/Controller.cpp:330]
libUnrealEditor-Engine.so!APawn::SpawnDefaultController() [/home/iplslam/UnrealEngine/Engine/Source/./Runtime/Engine/Private/Pawn.cpp:352]
libUnrealEditor-Engine.so!APawn::PostInitializeComponents() [/home/iplslam/UnrealEngine/Engine/Source/./Runtime/Engine/Private/Pawn.cpp:152]
libUnrealEditor-RapyutaSimulationPlugins.so!ARRBaseRobot::PostInitializeComponents() [/home/iplslam/turtlebot3-UE/Plugins/RapyutaSimulationPlugins/Source/RapyutaSimulationPlugins/Private/Robots/RRBaseRobot.cpp:170]
libUnrealEditor-Engine.so!ULevel::RouteActorInitialize(int) [/home/iplslam/UnrealEngine/Engine/Source/./Runtime/Engine/Private/Level.cpp:3294]
libUnrealEditor-Engine.so!UWorld::InitializeActorsForPlay(FURL const&, bool, FRegisterComponentContext*) [/home/iplslam/UnrealEngine/Engine/Source/./Runtime/Engine/Private/World.cpp:4768]
libUnrealEditor-Engine.so!UGameInstance::StartPlayInEditorGameInstance(ULocalPlayer*, FGameInstancePIEParameters const&) [/home/iplslam/UnrealEngine/Engine/Source/./Runtime/Engine/Private/GameInstance.cpp:508]
libUnrealEditor-UnrealEd.so!UEditorEngine::CreateInnerProcessPIEGameInstance(FRequestPlaySessionParams&, FGameInstancePIEParameters const&, int) [/home/iplslam/UnrealEngine/Engine/Source/./Editor/UnrealEd/Private/PlayLevel.cpp:3210]
libUnrealEditor-UnrealEd.so!UEditorEngine::OnLoginPIEComplete_Deferred(int, bool, FString, FPieLoginStruct) [/home/iplslam/UnrealEngine/Engine/Source/./Editor/UnrealEd/Private/PlayLevel.cpp:1672]
libUnrealEditor-UnrealEd.so!UEditorEngine::CreateNewPlayInEditorInstance(FRequestPlaySessionParams&, bool, EPlayNetMode) [/home/iplslam/UnrealEngine/Engine/Source/./Editor/UnrealEd/Private/PlayLevel.cpp:1932]
[... cut ...]
UnrealEditor!FEngineLoop::Tick() [/home/iplslam/UnrealEngine/Engine/Source/./Runtime/Launch/Private/LaunchEngineLoop.cpp:5363]
UnrealEditor!GuardedMain(char16_t const*) [/home/iplslam/UnrealEngine/Engine/Source/./Runtime/Launch/Private/Launch.cpp:204]
libUnrealEditor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/iplslam/UnrealEngine/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:269]
libc.so.6!__libc_start_main(+0xf2)
UnrealEditor!_start()
`
[image]

YuminosukeSato, 2023-06-10:
I did source /opt/ros/foxy/setup.bash before starting Unreal Editor.

yuokamoto, 2023-06-12:
Please run editor without source /opt/ros/foxy/setup.bash
Since rclUE has ros2 lib inside, you should not source /opt/ros/foxy/setup.bash

yuokamoto, 2023-06-12:
In your terminal which start navigation
- Please make sure that you execute ` cd turtlebot3-UE && source fastdds_setup.sh`
- try `ros2 topic echo /clock` if you can see /clock topic, ros2 communication is working.

YuminosukeSato, 2023-06-12:
I tryied try ros2 topic echo /clock.
I got "WARNING: topic [/clock] does not appear to be published yet
Could not determine the type for the passed topic".

yuokamoto, 2023-06-12:
Then I guess, something wrong with ros2 connection.
To connect to ros2 dds server, you need to execute `source fastdds_setup.sh` in the terminal

YuminosukeSato, 2023-06-12:
I  executed source fastdds_setup.sh in the terminal.
But ros2 topic echo /clock accured WARNING: topic [/clock] does not appear to be published yet Could not determine the type for the passed topic.
I excuteed ros2 topic list.
/parameter_events
/rosout
I don't see /clock.

yuokamoto, 2023-06-12:
hmm, not sure what is wrong.
here is the video of I run the turtlebot3-UE sim. Could you check?
https://github.com/rapyuta-robotics/turtlebot3-UE/assets/16463135/45aae6eb-4ea8-4890-a86a-08b4fa1932bf

YuminosukeSato, 2023-06-12:
navigation has moved. thank you.

closed 2023-06-12 by yuokamoto