rapyuta-robotics/turtlebot3-UE #66: How do I fix errors when packaging a Pixel Streaming project?

YuminosukeSato, 2023-07-03:
How do I fix errors when packaging a Pixel Streaming project?
https://docs.unrealengine.com/5.2/en-US/getting-started-with-pixel-streaming-in-unreal-engine/
Following the above, I got to 1.3 Package your Project for Windows. From the main menu in the Unreal Editor, choose Files > Package Project > Windows (64-bit).
But here I get an error.
```
UATHelper: Packaging (Linux): RunUAT ERROR: AutomationTool was unable to run successfully. Exited with code: 134
PackagingResults: Error: AutomationTool was unable to run successfully. Exited with code: 134
PackagingResults: Error: Unknown Error
```
How can this be resolved? Not packaged.
I was able to package it in the environment I made for testing simply to check if pixel streaming works. However, it could not be packaged in the environment of turtlebot3-UE.
Error log:
[UBT-UnrealPak-Linux-Development.txt](https://github.com/rapyuta-robotics/turtlebot3-UE/files/11933275/UBT-UnrealPak-Linux-Development.txt)

UE version:5.1.1
OS: ubuntu 20.04

yuokamoto, 2023-07-03:
Since currently rclUE supports only Linux, I guess, it is not feasible to packaging project for Windows.

james-yoo, 2023-07-03:
@YuminosukeSato,
Plugins(rclUE, RapyutaSimulationPlugin) are not work on Windows.

YuminosukeSato, 2023-07-05:
I use Ubuntu20.04.
I don't use Windows and wsl.
There is a site where Pixel Streaming can be done with unreal engine 5 of Ubuntu, and the package was possible even on Ubuntu. 
But turtlebot3-UE cannot be packaged.

YuminosukeSato, 2023-07-05:
https://tech.isid.co.jp/entry/unreal-engine-pixel-streaming
This sate use Ubuntu.
I think that Pixel Streaming plugin and Plugins (rclUE, RapyutaSimulationPlugin) may not be used at the same time even if the OS is UBUNTU.

yuokamoto, 2023-07-06:
- pixelstreaming is enabled in turtlebot3-UE already https://github.com/rapyuta-robotics/turtlebot3-UE/blob/devel/turtlebot3.uproject
- you can packaging turtlebot3-UE. if build error happen, it worth cleaning up Binaries/Intermeditate/Save folder

https://github.com/rapyuta-robotics/turtlebot3-UE/assets/16463135/2116a026-14bf-48d2-b8f0-10099dbeb691

YuminosukeSato, 2023-07-10:
Thank you. I was able to do it, can you navigate ros2 while doing pixel straming?

yuokamoto, 2023-07-10:
should be since pixelstreaming and navigation is totally independet

YuminosukeSato, 2023-07-11:
I have two questions for reference. Why can't I do ros2 navigation while pixel streaming? Other than pixel streaming, if I want to capture the camera attached to the turtle bot while using ros2 navigation, what are the alternatives?

yuokamoto, 2023-07-11:
- you can do navigation with pixel streaming.
- you can search other UE feature to capture image
- you can publish ROS 2 image topic by using RRROS2CameraComponent
https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/blob/bfac015d2227a3150f9c2b1d4b61c84c4fc6f19d/Source/RapyutaSimulationPlugins/Public/Sensors/RRROS2CameraComponent.h

YuminosukeSato, 2023-07-11:
I want to receive images from RRROS2CameraComponent while navigating and process the images with python opencv etc. (I want to take depth etc.) How should I do it?

yuokamoto, 2023-07-11:
it is already explained here
https://github.com/rapyuta-robotics/rclUE/issues/95#issuecomment-1595658591

YuminosukeSato, 2023-07-12:
I started Rviz on ROS2 and navigated, but the turtlebot of pixel streaming doesn't work.
export ROS_DISCOVERY_SERVER="127.0.0.1:8888"
export ROS_DISCOVERY_SERVER="127.0.0.1:11811"
Navigation didn't start with either setting.

yuokamoto, 2023-07-12:
I don't know how did you start pixel streaming but as always you should check one by one, e.g. move robot with cmd_vel, check scan, odom topic and etc.

YuminosukeSato, 2023-07-13:
https://tech.isid.co.jp/entry/unreal-engine-pixel-streaming

Done as above. First, set Additional Launch Parameters

From the menu "Edit" -> "Editor Preferences" -> "Level Editor" -> "Play" category,
Set the audio and IP/port by entering the following in Additional Launch Parameters:

AudioMixer -PixelStreamingIP=localhost -PixelStreamingPort=8888
Second, build the UE app for Linux

Build the project from the menu "Platforms" -> "Linux" -> "Package Project". Third, launch the WebRTC server.
Start "SignallingWebServer" (using Node.js-based Cirrus.js) provided by Unreal Engine.

set up:

cd ~/Desktop/UnrealEngine/Samples/PixelStreaming/WebServers/SignallingWebServer/platform_scripts/bash
chmod +x ./setup.sh
./setup.sh
Start Signaling Server:

chmod +x ./Start_SignallingServer.sh
./Start_SignallingServer.sh
Next, launch your UE application.

Go to the build directory and run the file with the project name ("turtlebot3-UE" in this case).
Optionally, specify the IP/port for headless launching of the UE app and Pixel Streaming.　　

cd ~/Desktop/Linux
chmod +x ./.turtlebot3-UE.sh
./turtlebot3-UE.sh -RenderOffScreen -PixelStreamingIP=127.0.0.1 -PixelStreamingPort=8888 Fourth, I ran the following in a new terminal: cd turtlebot3-UE && source fastdds_setup.sh #if you use ROS2 Discovery Server. You don't need this if you start editor with ./run_editor false.
ros2 launch nav2_bringup tb3_simulation_launch.py ​​use_simulator:=False map:=<path to turtlebot3-UE>/Content/Turtlebot3_benchmark.yaml

yuokamoto, 2023-07-13:
As I said you should do one by one and separate issue. 
- you are trying to do both pixel streaming and run ros2 application with packaged UE project. You should do separately. As I said before pixel streaming and ROS2 are completely independent features. you should test one by one.
- you should test low level api first if something not work, i.e. check robot moves with cmd_vel and scan, odom topics are there or not.

yuokamoto, 2023-08-12:
closed because no activity

closed 2023-08-12 by yuokamoto