rapyuta-robotics/turtlebot3-UE #67: How do I set the camera installed on my turtolbot3 as the first camera in play?

YuminosukeSato, 2023-07-10:
I want to designate the first camera in play as the camera installed in turtolbot3 in unreal engine 5. However, I have to change the blue print and I don't know which blue print to change. How do I set the camera installed on my turtolbot3 as the first camera in play?

yuokamoto, 2023-07-10:
please check
https://docs.unrealengine.com/4.26/ja/InteractiveExperiences/UsingCameras/SwitchingCameras/

YuminosukeSato, 2023-07-12:
I installed the camera, but the camera actor does not appear in the blueprint. And I have to change the blueprint, but I want to connect to ROS2, so when I'm playing while maintaining this state
I want to connect to turtlebot's camera.

yuokamoto, 2023-07-12:
you can assign any actor which have camera component as a target

[image]

this is more general question about UE and not related to this repo. you can search more on UE doc and forum.
https://forums.unrealengine.com/t/set-view-target-from-camera-component/311905