rapyuta-robotics/turtlebot3-UE #70: Invalid frame ID + Lookup would require extrapolation into the future... Navigation not working in rviz2

alexzhang13, 2023-08-01:
I'm trying to get the Turtlebot example working on my machine. Currently using UE5.1 on Ubuntu 22.04 w/ ROS2 Humble. I have checked out the corresponding branches (Jammy, devel-humble, etc.) for this repository and the sub-repository plugins. 

In UE5, I can see the LiDAR scans, but I can't get the robot to move with rviz2. After putting in the initial pose estimate, I can't actually get the robot to navigate and move around. I've also tested, and I'm able to get the robot to move in UE5 with teleop control from ROS2. Checking the logs, this is what gets returned constantly from rviz:

```
[rviz2-3] [ERROR] [1690931335.516706034] [rviz2]: Lookup would require extrapolation into the future.  Requested time 683.613197 but the latest data is at time 683.589966, when looking up transform from frame [base_scan] to frame [map]
[rviz2-3] Warning: Invalid frame ID "wheel_left_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 93 in ./src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "wheel_right_link" passed to canTransform argument source_frame - frame does not exist
...
```
This is what the rviz displays look like. 
[image]

I'm new to ROS2 and rviz, so I was wondering if I was doing something wrong here, or if this was a bug. Thanks.

yuokamoto, 2023-08-02:
- attached video to run tb3 nav
- missing tf of wheels shouldn't be blocker of navigation.
https://github.com/rapyuta-robotics/turtlebot3-UE/assets/16463135/2daed39c-448d-40a1-919d-cd534838a13a

james-yoo, 2023-08-02:
@alexzhang13 
You should check TF between 'base_scan' and 'map'. There is missing link between them.

alexzhang13, 2023-08-02:
@james-yoo Is this something I have to explicitly define? I'm basically just loading in their example and trying to replicate their steps.

james-yoo, 2023-08-03:
@alexzhang13 
Your Nav 2 stack was not loaded successfully as below. This may cause the wrong transform as you mentioned above
[image]

alexzhang13, 2023-08-03:
Gotcha thanks, I'll go back and see what might be going wrong here in my installation setup.

closed 2023-08-03 by alexzhang13