rapyuta-robotics/turtlebot3-UE #72: problem in installing rclUE plugin in UE5 trying "Turtlebot3-UE"

MacBuki, 2023-08-10:
I am learning about connecting UE5 and ROS2 humble.And, I've searched and found your solution, rclUE.But, from the beginning, I got stuck on adding plugin rclUE to unreal.I'm using ubuntu 22.04, UE 5.2.1, ROS2 Humble.As I add the 'rclUE' folder to "/Linux_Unreal_Engine_5.2.1/Engine/Plugins/", It got crashed.Is there another way to add plugin to unreal? Also, before trying "Turtlebot3-UE", What other plugins or setting do I need??

yuokamoto, 2023-08-10:
you don't need to put project-level plugin to UnrealEngine directory.
you just need to put rclUE under Plugins folder of project like turtlebot3-UE.

Currently our plugins supports UE 5.1 not 5.2 but there maybe a working branch for UE 5.2 cc @Tadinu

james-yoo, 2023-08-10:
@MacBuki 
I confirm that using the 'UE5.2' branch perfectly works on Ubuntu 22.04, UE5.2.1 and ROS 2 humble.

MacBuki, 2023-08-10:
@yuokamoto Okay, succeed opening "Turtlebot3-UE".
UE said I have to copy the project, because, the version is not mach. so, I did it.
however, I got tons of errors....
what I got is:
"Turtlebot3-UE"-Jammy
"rclUE"-UE5
"RapyutaSimulationPlugins"-UE5.2

Is the branch what I got is work?

**errors..**
> The summary for the package '/RapyutaSimulationPlugins/Robots/Turtlebot3/Kinematics/BP_TurtlebotBurgerVehicle' is invalid. Check that the file is of the expected type and not corrupted.
> The summary for the package '/RapyutaSimulationPlugins/Robots/Turtlebot3/Kinematics/BP_TurtlebotWaffleVehicle' is invalid. Check that the file is of the expected type and not corrupted.
> The summary for the package '/RapyutaSimulationPlugins/Robots/Turtlebot3/Physics/BP_TurtlebotBurger' is invalid. Check that the file is of the expected type and not corrupted.
> The summary for the package '/RapyutaSimulationPlugins/Robots/Turtlebot3/Physics/BP_TurtlebotWaffle' is invalid. Check that the file is of the expected type and not corrupted.

james-yoo, 2023-08-10:
@MacBuki 
You should use 'UE5_devel_humble' branch for rclUE plugin

MacBuki, 2023-08-11:
Okay, trying with
-"Turtlebot3-UE"-Jammy
-"rclUE"-UE5_devel_humble
-"RapyutaSimulationPlugins"-UE5.2

No, crach on playing! yeah!

the example I want to try is "RobotArmExample"
But I think the model is missing?
[image]
[image]

> 
> Could not find a variable named "ElbowJoint" in 'RobotArmExample'.
> Make sure 'RobotArmExample' has been compiled for  Get ElbowJoint
> Could not find a variable named "ShoulderLiftJoint" in 'RobotArmExample'.
> Make sure 'RobotArmExample' has been compiled for  Get ShoulderLiftJoint
> Could not find a variable named "ShoulderPanJoint" in 'RobotArmExample'.
> Make sure 'RobotArmExample' has been compiled for  Get ShoulderPanJoint
> Node  Unknown  is not referencing a valid level actor
> The property associated with  Elbow Joint  could not be found in '/Game/Maps/RobotArmExample.RobotArmExample_C'
> The property associated with  Shoulder Lift Joint  could not be found in '/Game/Maps/RobotArmExample.RobotArmExample_C'
> The property associated with  Shoulder Pan Joint  could not be found in '/Game/Maps/RobotArmExample.RobotArmExample_C'

yuokamoto, 2023-08-12:
Since UE5.2 branch is created before robot arm examples are added.
I created branch with rebase from devel. please try it 
https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/tree/UE5.2_rebase

MacBuki, 2023-08-17:
Okay, trying with
-"Turtlebot3-UE"-Jammy
-"rclUE"-UE5_devel_humble
-"RapyutaSimulationPlugins"-UE5.2_rebase

I've checked that the file is invalid?
[image]
[image]

>  The summary for the package '/RapyutaSimulationPlugins/Robots/SampleArm/BP_KinematicSimpleArm' is invalid. Check that the file is of the expected type and not corrupted.
> The summary for the package '/RapyutaSimulationPlugins/Robots/UR10/BP_KinematicUR10' is invalid. Check that the file is of the expected type and not corrupted.
> The summary for the package '/RapyutaSimulationPlugins/Robots/UR10/BP_PhysicsUR10' is invalid. Check that the file is of the expected type and not corrupted.

yuokamoto, 2023-08-17:
Did you git lfs pull? it maybe not pulled properly.

yuokamoto, 2023-09-16:
Please reopen if issue still exists

closed 2023-09-16 by yuokamoto