rapyuta-robotics/turtlebot3-UE #76: Query Regarding Controlling a Turtlebot’s Camera in ROS2 Humble

Buddies-as-you-know, 2023-11-12:
Hello,

I am working with ROS2 Humble and am looking for guidance on controlling the camera on a Turtlebot. I previously came across a repository that was compatible with ROS2 Foxy, but it seems it’s not suitable for Humble. Could you please provide any information or resources on how to control the Turtlebot’s camera specifically in the Humble version? Advice on any relevant documentation, tutorials, or alternative repositories compatible with Humble would be incredibly helpful.

Thank you in advance for your time and assistance.

Best regards,

yuokamoto, 2023-11-13:
Thanks for your question.

1. There is jammy branch which can run on Ubuntu22.04 with humble. https://github.com/rapyuta-robotics/turtlebot3-UE/tree/jammy 
2. this video in discussion helps to attach camera to turtlebot I think. https://github.com/rapyuta-robotics/rclUE/issues/95#issuecomment-1600582298

Buddies-as-you-know, 2023-11-13:
Thanks for your reply.
How can I record camera images?

yuokamoto, 2023-11-13:
I think there are two options
1. UE's feature to record from camera(I don't have much experience) https://docs.unrealengine.com/4.26/ja/AnimatingObjects/Sequencer/Workflow/RenderAndExport/RenderMovies/
2. ROS's feature to record by `ros2 bag record` https://docs.ros.org/en/foxy/Tutorials/Beginner-CLI-Tools/Recording-And-Playing-Back-Data/Recording-And-Playing-Back-Data.html

yuokamoto, 2023-11-21:
For clarification and my understanding, are there any more blocker?

Buddies-as-you-know, 2023-11-21:
https://docs.google.com/presentation/d/1eZNXPmLtfYkiXLVbxpxOCY--ukxQFXVAd3Uaja0tKRQ/edit#slide=id.g1f425351010_0_31
I want to run yolo in real time as shown above, but it doesn't work because it doesn't support foxy. Is there an alternative?

yuokamoto, 2023-11-21:
There is a rclUE and turtlebot3-UE branch for humble as well
https://github.com/rapyuta-robotics/turtlebot3-UE/issues/76#issuecomment-1807289366

Buddies-as-you-know, 2023-11-21:
[image]
I'm running YOLO, but for some reason it stops at this screen and the video doesn't display.

yuokamoto, 2023-11-21:
Ah, I see what your point now sorry.
Since it is yolo issue, it is better to ask in different place I think. I'm not familliar with yolo

james-yoo, 2023-11-21:
@Buddies-as-you-know 
You should change the image topic name to '/image_raw' as below:
https://docs.google.com/presentation/d/1eZNXPmLtfYkiXLVbxpxOCY--ukxQFXVAd3Uaja0tKRQ/edit#slide=id.g1b671e8ccb4_0_11

Also, you have to change main.py to publish result image(e.g bounding boxe image) as described below:
https://docs.google.com/presentation/d/1eZNXPmLtfYkiXLVbxpxOCY--ukxQFXVAd3Uaja0tKRQ/edit#slide=id.g1f6a6b86cd8_0_21

Buddies-as-you-know, 2023-11-22:
I changed the image topic name to '/image_raw' as below:
[image]
I changed main.py.
[image]
But,  the video doesn't display.

Buddies-as-you-know, 2024-01-08:
Thank　you. I did this.