rapyuta-robotics/turtlebot3-UE #82: Packaging error

james-yoo, 2024-03-19:
Hi
I've got following error while packaging turtlebot3-UE for Linux.
```
UATHelper: Packaging (Linux): /home/user/turtlebot3-UE/Plugins/RapyutaSimulationPlugins/Source/RapyutaSimulationPlugins/Private/Drives/RRFloatingMovementComponent.cpp:367:77: error: no member named 'IsDebugEnabled' in 'UAvoidanceManager'
UATHelper: Packaging (Linux):     const bool bRVODebugEnabled = RAPYUTA_SIM_DEBUG && RVOAvoidanceManager->IsDebugEnabled(RVOAvoidanceUID);
```
Packaging was conducted under following condition:
- turtlebot3-UE (branch: jammy)
- RapyutaSimulationPlugins(branch: UE5.3)
- rclUE(branch: UE5_devel_humble)

yuokamoto, 2024-03-19:
I update UE5.3(accidentaly without PR)
Could you please try?

james-yoo, 2024-03-20:
@yuokamoto 
Thanks for your quick response. Issue resolved.

james-yoo, 2024-03-21:
@yuokamoto 
Re-opened issue. The [commit](https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/commit/2c3087021b6b60ab534755baeab94995945f26d0) still giving same error.
I didn't get the error because I commented out the following line when I packaged it.
```
const bool bRVODebugEnabled = false; //RAPYUTA_SIM_DEBUG && RVOAvoidanceManager->IsDebugEnabled(RVOAvoidanceUID);
```

yuokamoto, 2024-03-23:
I'm using this commit 728f3add2085f26e0e80438ae7a3674ae8af2d70

and also I was able to pcakge with jammy_UE5.3 branch
https://github.com/rapyuta-robotics/turtlebot3-UE/tree/jammy_UE5.3

closed 2025-02-03 by james-yoo