rapyuta-robotics/turtlebot3-UE #83: Robot Slightly turning to the right.

dwhite18, 2024-03-20:
I've noticed when controlling via teleop_twist_keyboard the robot and giving a command to go purely forward the robot will slowly arc towards the right. I've check the inputs to the physics components for the wheels during runtime and the values are correct with the right being the negative of the left. These are the target angular velocities for the wheels.

Second I noticed that the robot moves even when it is supposed to be stationary. 

What could be causing these?

james-yoo, 2024-03-20:
Same issue here

dwhite18, 2024-03-21:
I'm getting about 1 degree per meter with drift by setting the friction lower for the physics material for the wheels. When the friction is higher I'm seeing a much greater increase in drift.

dwhite18, 2024-03-21:
I found that if I flip the rotation of the right wheel physics component and also flip the sign on the right wheel velocity calculation in the differential drive the drift practically disappears.

yuokamoto, 2024-03-22:
Thanks for reporting.

Are you using BP_RRTurtlebotWaffle?
I'm investigating but it seems BP_RRTurtlebotWaffle does large drafting but BP_TurtlebotWaffle has less.

yuokamoto, 2024-03-23:
[image]

I updpate Angular Dumper and Angular Sprint to avoid drifting of the robot.
Robot still does not always move straight. need more investigation.

https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/pull/275

james-yoo, 2024-04-30:
@yuokamoto 
Any updates? I am still struggling with this issue.

yuokamoto, 2024-04-30:
Sorry, I haven't found solution.
Could you use BP_TurtlebotWaffle instead of BP_RRTurtlebotWaffle? BP_TurtlebotWaffle seems have less drifiting.

yuokamoto, 2024-05-05:
I created PR.
not perfect but at least able to avoid driting
https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/pull/278

james-yoo, 2024-05-05:
@yuokamoto 
Could you please explain what you fixed?

yuokamoto, 2024-05-06:
I updated BP_RRTurtlebotWaffles' physics joint's(BaseWheelRight and BaseWheelLeft) damper and force limit to become same as BP_TurtlebotWaffle

original
[image]

new
[image]