rapyuta-robotics/turtlebot3-UE #84: spawn all asset is not working.

Kim-YoungJ, 2024-04-12:
I am using ubuntu 22.04 (jammy) and Unreal Engine 5.1.1. i follow description in readme.  Then working fine world map which i use turtlebot3 benchmark. However, the turtlebot3 does not spawn it. In Unreal engine log massage says that BP_TurtlebotBuger is invalid. take a look picture. 
What should i do? can you help this problem?
[image]

yuokamoto, 2024-04-12:
Hello,
- Are you using `jammy` branch of turtlebot3-UE? Could you try `git submodule update --init --recursive`?
- Did you do git-lfs pull? This can happen if large binary are not pulled.

Kim-YoungJ, 2024-04-15:
@yuokamoto  Thanks a lot. i solved it using git-lfs pull, the large binary files are not pulled as you mentioned.

closed 2024-04-15 by yuokamoto