rapyuta-robotics/turtlebot3-UE #88: Turtlebot topics not visible, only on Ubuntu 22.04 jammy

pabsan-0, 2024-07-05:
Hi, I can't get the simulation and terminals to communicate using source run_discovery_service.sh in Ubuntu 22.04 + humble, using the jammy branch and UE5.1. After starting the simulation, I can't see topics published by the engine. Here are my outputs:

- Editor
```
~/dev/turtlebot3-UE$ ./run_editor.sh 
2024-07-05 11:26:38.919 [XMLPARSER Error] Found incorrect tag 'userTransports' -> Function fillDataNode
2024-07-05 11:26:38.919 [XMLPARSER Error] Error parsing participant profile -> Function parseXMLParticipantProf
2024-07-05 11:26:38.920 [XMLPARSER Error] Error parsing profile's tag participant -> Function parseProfiles
2024-07-05 11:26:38.920 [XMLPARSER Error] Error parsing '/home/catec/dev/turtlebot3-UE/fastdds_config.xml' -> Function loadXMLFile
### Server is running ###
  Participant Type:   SERVER
  Security:           NO
  Server ID:          0
  Server GUID prefix: 44.53.00.5f.45.50.52.4f.53.49.4d.41
  Server Addresses:   UDPv4:[0.0.0.0]:11811
The daemon has been started
Increasing per-process limit of core file size to infinity.
* UE5 output * 
```

- Listener 
```
~/dev/turtlebot3-UE$ source run_discovery_service.sh 
2024-07-05 11:26:56.890 [XMLPARSER Error] Found incorrect tag 'userTransports' -> Function fillDataNode
2024-07-05 11:26:56.890 [XMLPARSER Error] Error parsing participant profile -> Function parseXMLParticipantProf
2024-07-05 11:26:56.890 [XMLPARSER Error] Error parsing profile's tag participant -> Function parseProfiles
2024-07-05 11:26:56.890 [XMLPARSER Error] Error parsing '/home/catec/dev/turtlebot3-UE/fastdds_config.xml' -> Function loadXMLFile
2024-07-05 11:26:56.892 [RTPS_PARTICIPANT Error] Discovery Server wasn't able to allocate the specified listening port. -> Function createParticipant
2024-07-05 11:26:56.893 [DOMAIN_PARTICIPANT Error] Problem creating RTPSParticipant -> Function enable
Server creation failed with the given settings. Please review locators setup.
The daemon is already running
fastdds discovery <discovery-args>

1
 fast-discovery-server tool not found!
~/dev/turtlebot3-UE$ ros2 topic list
/parameter_events
/rosout
~/dev/turtlebot3-UE$ 
```

This was not an issue in Ubuntu 20.04 + foxy under the devel branch. I have noticed that the fastdds_config.xml is the same for both devel and jammy. Might this be an issue with this config file, which is outdated for the more recent humble fastdds version?

As a workaround, not using the repo scripts and just running `$UE5_DIR/Engine/Binaries/Linux/UnrealEditor turtlebot.uproject` then sourcing ROS on the receiving terminal makes all topics visible and `echo`able.

mhl787156, 2024-07-19:
Hi @pabsan-0 I'm also encountering this problem trying to test locally! Did running `./run_editor.sh false` with the discovery server turned off work for you? Because it also doesnt seem to for me - I can't seem to get any ros topics in another terminal. 

I was just reading and it seems there's issues running discovery server on localhost: https://github.com/eProsima/Fast-DDS/issues/2031#issuecomment-1176388874

mhl787156, 2024-07-23:
Hi after some testing I stumbled upon the following post which pointed towards a solution! https://github.com/ros2/rmw_fastrtps/issues/676#issuecomment-1470432237 

Essentially discovery server by default will work in the background, but all of the ros2 cli tools fail. They require the super client settings in order for the ros2 cli tools to work (https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/discovery_server/ros2_discovery_server.html#daemon-s-related-commands). This combined with an outdated format for the fastdds config meant that you could not get a connection and instead got the Error parsing errors. 

What worked for me was to update the fastdds_config.xml to the following:
```
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
  <transport_descriptors>
    <transport_descriptor>
      <transport_id>udp_transport</transport_id>
      <type>UDPv4</type>
      <maxMessageSize>1400</maxMessageSize>
    </transport_descriptor>
  </transport_descriptors>

  <participant profile_name="participant_profile" is_default_profile="true">
    <rtps>
      <userTransports>
        <transport_id>udp_transport</transport_id>
      </userTransports>
      <useBuiltinTransports>false</useBuiltinTransports>

      <builtin>
        <discovery_config>
          <discoveryProtocol>SUPER_CLIENT</discoveryProtocol>
          <discoveryServersList>
            <RemoteServer prefix="44.53.00.5f.45.50.52.4f.53.49.4d.41">
              <metatrafficUnicastLocatorList>
                <locator>
                  <udpv4>
                    <address>127.0.0.1</address>
                    <port>11811</port>
                  </udpv4>
                </locator>
              </metatrafficUnicastLocatorList>
            </RemoteServer>
          </discoveryServersList>
        </discovery_config>
      </builtin>
    </rtps>
  </participant>
</profiles>
```
> The docs mention that the `RemoteServer` should be set to whatever the output of `fastdds discovery -i 0` is. 

Then additionally you should add the following to the bottom of `fastdds_setup.sh` 
```
# Restart the daemon to setup the super client config
echo "Restarting ROS2 Daemon For New FASTRTPS Settings" 
ros2 daemon stop
ros2 daemon start
```

Which restarts the ros2 daemon with the super client enabled automatically! 

Hopefully this works for you!

mhl787156, 2024-07-25:
Edit! 

In `run_discovery_service.sh` you may want to change the call to fastdds discovery into a subshell rather than background process. This is so that the discovery server also stops when you exit/ctrl c, so that the next time you run, the discovery server restarts correctly. 

```
#!/bin/bash

source fastdds_setup.sh
source /opt/ros/humble/setup.bash
(exec fastdds discovery -i 0) # Run as sub shell so ctrl c will stop it
ros2 daemon start
```

I have had issues with the discovery server not reconnecting properly with a restarted UE5 instance.