rapyuta-robotics/turtlebot3-UE #89: tf from rclUE has prefix

KMerdan, 2024-09-29:
## Description

I am following the documentation to run the `jammy` branch of `turtlebot3-UE`. However, the `turtlebot3-UE` publisher automatically adds the `BP_TurtlebotWaffle` prefix to the `frame_id` & `topic_name` of the published topics. For example:

- `/BP_TurtlebotWaffle/odom`
- `/BP_TurtlebotWaffle/base_scan`
- `/BP_TurtlebotWaffle/base_link`

This is causing issues because `nav2` expects these frames without the prefix (e.g., `odom`, `base_scan`, `base_link`).

## Expected Behavior

I would like to know how to remove or customize the `BP_TurtlebotWaffle` prefix in `turtlebot3-UE` so that the frame IDs are published without it.

For example:

- `/odom` instead of `/BP_TurtlebotWaffle/odom`
- `/base_scan` instead of `/BP_TurtlebotWaffle/base_scan`
- `/base_link` instead of `/BP_TurtlebotWaffle/base_link`

## Additional Information

I am not very familiar with Unreal Engine Blueprints and have mainly been looking at the code. However, I haven't found anything in the source code that directly relates to this issue. It would be helpful if someone could point me in the right direction or suggest how to modify this behavior.

yuokamoto, 2024-10-02:
Hi,
Thanks for you comment ans sorry for late response.

Could you try
- setting bUseActorNameAsNamespace = false
https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/blob/9dce510b48a16122855d48c18032741708a305d3/Source/RapyutaSimulationPlugins/Public/Robots/RRBaseROS2Interface.h#L75

KMerdan, 2024-10-02:
Thanks, that works.
a little bit off the topic, i still have trouble to run the turtleBot3-UE example with nav2 stack like this:
`ros2 launch nav2_bringup tb3_simulation_launch.py use_simulator:=False map:=./Content/Turtlebot3_benchmark.yaml`
where the tf between `map` and `odom` are not been published, 
[image]
so it results the rviz result like this
[image]

yuokamoto, 2024-10-02:
I'll check later
Did you provide initial pose?

KMerdan, 2024-10-02:
yes, i did.

james-yoo, 2024-10-02:
@KMerdan
I think 'wheel_right' and 'wheel_left' link should be attached to 'base_link' not 'base'.

KMerdan, 2024-10-02:
@james-yoo i think you are right. but this is what `turtleBot3-UE` default publish result, i didn't do any modification yet, just try to run it.

yuokamoto, 2024-10-08:
Sorry for late
- I updated jammy branch. Could you try?
- I attached video with the branch.

https://github.com/user-attachments/assets/137b0ac9-d8a8-424a-96bd-4d171d4886e0

[frames_2024-10-08_23.26.27.pdf](https://github.com/user-attachments/files/17294891/frames_2024-10-08_23.26.27.pdf)

KMerdan, 2024-10-10:
Confirmed here, works like a charm. Thanks.
@yuokamoto

yuokamoto, 2024-10-11:
Thanks for your reporting

closed 2024-10-11 by yuokamoto