rapyuta-robotics/turtlebot3-UE #91: Error: arithmetic between different enumeration types

arjuntheprogrammer, 2025-01-10:
Ubuntu 22.04.2 LTS
ROS2 Humble
UE 5.1
Git Branch: Humble

Got this following error while running the command: `./run_editor.sh true $(pwd)`

Partial ERROR LOG
```
4/47] Compile SharedPCH.Engine.ShadowErrors.Latest.h
In file included from /home/ubuntu/Development/turtlebot3-UE/Intermediate/Build/Linux/B4D820EA/turtlebot3Editor/Development/Engine/SharedPCH.Engine.ShadowErrors.Latest.h:368:
In file included from ./Runtime/Engine/Public/EngineSharedPCH.h:216:
/home/ubuntu/UE5.1/Engine/Source/Runtime/Core/Public/Math/SHMath.h:39:38: error: arithmetic between different enumeration types ('TSHVector<2>::(unnamed enum at Runtime/Core/Public/Math/SHMath.h:37:2)' and 'TSHVector<2>::(unnamed enum at Runtime/Core/Public/Math/SHMath.h:38:2)') is deprecated [-Werror,-Wdeprecated-anon-enum-enum-conversion]
        enum { NumSIMDVectors = (MaxSHBasis + NumComponentsPerSIMDVector - 1) / NumComponentsPerSIMDVector };
                                 ~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ubuntu/UE5.1/Engine/Source/Runtime/Core/Public/Math/SHMath.h:460:21: note: in instantiation of template class 'TSHVector<2>' requested here
inline TSHVector<2> TSHVector<2>::SHBasisFunction(const FVector& Vector)
                    ^
/home/ubuntu/UE5.1/Engine/Source/Runtime/Core/Public/Math/SHMath.h:40:41: error: arithmetic between different enumeration types ('TSHVector<2>::(unnamed enum at Runtime/Core/Public/Math/SHMath.h:39:2)' and 'TSHVector<2>::(unnamed enum at Runtime/Core/Public/Math/SHMath.h:38:2)') is deprecated [-Werror,-Wdeprecated-anon-enum-enum-conversion]
        enum { NumTotalFloats = NumSIMDVectors * NumComponentsPerSIMDVector };
                                ~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ubuntu/UE5.1/Engine/Source/Runtime/Core/Public/Math/SHMath.h:39:38: error: arithmetic between different enumeration types ('TSHVector<3>::(unnamed enum at Runtime/Core/Public/Math/SHMath.h:37:2)' and 'TSHVector<3>::(unnamed enum at Runtime/Core/Public/Math/SHMath.h:38:2)') is deprecated [-Werror,-Wdeprecated-anon-enum-enum-conversion]
        enum { NumSIMDVectors = (MaxSHBasis + NumComponentsPerSIMDVector - 1) / NumComponentsPerSIMDVector };
                                 ~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ubuntu/UE5.1/Engine/Source/Runtime/Core/Public/Math/SHMath.h:472:21: note: in instantiation of template class 'TSHVector<3>' requested here
inline TSHVector<3> TSHVector<3>::SHBasisFunction(const FVector& Vector)
                    ^
/home/ubuntu/UE5.1/Engine/Source/Runtime/Core/Public/Math/SHMath.h:40:41: error: arithmetic between different enumeration types ('TSHVector<3>::(unnamed enum at Runtime/Core/Public/Math/SHMath.h:39:2)' and 'TSHVector<3>::(unnamed enum at Runtime/Core/Public/Math/SHMath.h:38:2)') is deprecated [-Werror,-Wdeprecated-anon-enum-enum-conversion]
        enum { NumTotalFloats = NumSIMDVectors * NumComponentsPerSIMDVector };
                                ~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/ubuntu/Development/turtlebot3-UE/Intermediate/Build/Linux/B4D820EA/turtlebot3Editor/Development/Engine/SharedPCH.Engine.ShadowErrors.Latest.h:368:
In file included from ./Runtime/Engine/Public/EngineSharedPCH.h:507:
/home/ubuntu/UE5.1/Engine/Source/Runtime/Engine/Public/PrimitiveUniformShaderParameters.h:543:64: error: comparison of different enumeration types ('FPrimitiveSceneShaderData::(unnamed enum at Runtime/Engine/Public/PrimitiveUniformShaderParameters.h:536:2)' and 'FScatterUploadBuffer::(unnamed enum at Runtime/RenderCore/Public/UnifiedBuffer.h:71:2)') is deprecated [-Werror,-Wdeprecated-anon-enum-enum-conversion]
                static_assert(FPrimitiveSceneShaderData::DataStrideInFloat4s == FScatterUploadBuffer::PrimitiveDataStrideInFloat4s,"");
                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/ubuntu/Development/turtlebot3-UE/Intermediate/Build/Linux/B4D820EA/turtlebot3Editor/Development/Engine/SharedPCH.Engine.ShadowErrors.Latest.h:368:
In file included from ./Runtime/Engine/Public/EngineSharedPCH.h:608:
/home/ubuntu/UE5.1/Engine/Source/Runtime/Engine/Classes/Engine/NetConnection.h:75:74: error: arithmetic between different enumeration types ('(unnamed enum at Runtime/Engine/Classes/Engine/NetConnection.h:73:1)' and '(unnamed enum at Runtime/Engine/Classes/Engine/NetConnection.h:74:1)') is deprecated [-Werror,-Wdeprecated-anon-enum-enum-conversion]
enum { MAX_PACKET_HEADER_BITS = MAX_PACKET_RELIABLE_SEQUENCE_HEADER_BITS + MAX_PACKET_INFO_HEADER_BITS  };
                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~
6 errors generated.
```

FULL ERROR LOG:

```
(base) ➜  turtlebot3-UE git:(humble) ✗ ./run_editor.sh true $(pwd)  
2025-01-10 21:22:10.397 [XMLPARSER Error] Found incorrect tag 'userTransports' -> Function fillDataNode
2025-01-10 21:22:10.397 [XMLPARSER Error] Error parsing participant profile -> Function parseXMLParticipantProf
2025-01-10 21:22:10.397 [XMLPARSER Error] Error parsing profile's tag participant -> Function parseProfiles
...
...
Parsing headers for turtlebot3Editor
  Running Internal UnrealHeaderTool /home/ubuntu/Development/turtlebot3-UE/turtlebot3.uproject /home/ubuntu/Development/turtlebot3-UE/Intermediate/Build/Linux/B4D820EA/turtlebot3Editor/Development/turtlebot3Editor.uhtmanifest -WarningsAsErrors -installed
Total of 16 written
Reflection code generated for turtlebot3Editor in 1.0307195 seconds
@progress pop
------- Build details --------
Using toolchain located at '/home/ubuntu/UE5.1/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v20_clang-13.0.1-centos7/x86_64-unknown-linux-gnu'.
Using clang (/home/ubuntu/UE5.1/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v20_clang-13.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang++) version 'clang version 13.0.1
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /home/ubuntu/UE5.1/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v20_clang-13.0.1-centos7/x86_64-unknown-linux-gnu/bin' (string), 13 (major), 0 (minor), 1 (patch)
Using bundled libc++ standard C++ library.
Using lld linker
Using llvm-ar (/home/ubuntu/UE5.1/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v20_clang-13.0.1-centos7/x86_64-unknown-linux-gnu/bin/llvm-ar) version 'LLVM (http://llvm.org/):
  LLVM version 13.0.1
  Optimized build.
  Default target: x86_64-unknown-linux-gnu
  Host CPU: znver3 (string)'
[... 36 lines cut ...]
In file included from ./Runtime/Engine/Public/EngineSharedPCH.h:608:
/home/ubuntu/UE5.1/Engine/Source/Runtime/Engine/Classes/Engine/NetConnection.h:75:74: error: arithmetic between different enumeration types ('(unnamed enum at Runtime/Engine/Classes/Engine/NetConnection.h:73:1)' and '(unnamed enum at Runtime/Engine/Classes/Engine/NetConnection.h:74:1)') is deprecated [-Werror,-Wdeprecated-anon-enum-enum-conversion]
enum { MAX_PACKET_HEADER_BITS = MAX_PACKET_RELIABLE_SEQUENCE_HEADER_BITS + MAX_PACKET_INFO_HEADER_BITS  };
                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~
6 errors generated.
[5/47] Compile Module.rclUE.5_of_12.cpp cancelled
[6/47] Compile Module.rclUE.12_of_12.cpp cancelled
[7/47] Compile Module.rclUE.4_of_12.cpp cancelled
...
[44/47] Compile Module.RapyutaSimulationPlugins.8_of_9.cpp
[45/47] Link (lld) libUnrealEditor-RapyutaSimulationPlugins.so cancelled
[46/47] Link (lld) libUnrealEditor-turtlebot3.so cancelled
[47/47] WriteMetadata turtlebot3Editor.target cancelled
LogInit: Warning: Still incompatible or missing module: turtlebot3
LogInit: Warning: Still incompatible or missing module: RapyutaSimulationPlugins
LogInit: Warning: Still incompatible or missing module: rclUE
LogCore: Engine exit requested (reason: EngineExit() was called)
LogExit: Preparing to exit.
LogVirtualization: UE::Virtualization was shutdown
...
LogExit: Exiting.
LogInit: Tearing down SDL.
Exiting abnormally (error code: 1)
(base) ➜  turtlebot3-UE git:(humble) ✗ 
```

SuribaCN, 2025-02-08:
As a temporary solution, you can add `#pragma clang diagnostic ignored "-Wdeprecated-anon-enum-enum-conversion"` at the beginning of `SHMath.h` and recompile the engine from source.It just worked. The cause of the error can be found at https://gcc.gnu.org/onlinedocs/gcc/C_002b_002b-Dialect-Options.html#C_002b_002b-Dialect-Options.

For reference, here are the methods I tried but failed:

1. Added `CppStandard = CppStandardVersion.Cpp17;` in `rclUE.Build.cs`, `RapyutaSimulationPlugins.Build.cs`, `turtlebot3.Build.cs`, and `Core.Build.cs`. In theory, even without this line, UE 5.1 should be compiled with the C++17 standard.
2. Passed parameters directly after `make`.
3. Added `Arguments.Add("-Wno-deprecated-enum-enum-conversion");` and `Arguments.Add("-Wno-undefined-bool-conversion");` in functions like `GetCompileArguments_WarningsAndErrors()`/`GetCompileArguments_Global` in `LinuxToolChain.cs`/`ClangToolChain.cs` and recompiled UBT.

Ubuntu 22.04.04 LTS  
ROS2 Humble  
UE 5.1 built from source  
Using the `jammy` branch of `turtlebot3-UE` and manually replaced the `rclUE` plugin with the `UE5_devel_humble` branch (commit `1f84286`).

yuokamoto, 2025-02-09:
Could you try jammy_UE5.1 branch and update submodules?