rapyuta-robotics/turtlebot3-UE #92: Symbol lookup error: /opt/ros/humble/lib/librmw_fastrtps_shared_cpp.so: undefined symbol: rmw_time_equal

arjuntheprogrammer, 2025-01-10:
Ubuntu 22.04.2 LTS
ROS2 Humble
UE 5.1
Git Branch: devel

UE crashed with the following error on hitting the play button in UE Editor after running `./run_editor.sh true $(pwd)` command:

```
[2025.01.10-16.12.11:629][815]LogInit: FAudioDevice initialized.
[2025.01.10-16.12.11:629][815]LogAudio: Display: Audio Device (ID: 2) registered with world 'Turtlebot3_benchmark'.
[2025.01.10-16.12.11:631][815]LogLoad: Game class is 'RRROS2GameMode'
[2025.01.10-16.12.11:632][815]LogWorld: Bringing World /Game/Maps/UEDPIE_0_Turtlebot3_benchmark.Turtlebot3_benchmark up for play (max tick rate 0) at 2025.01.10-21.42.11
[2025.01.10-16.12.11:632][815]LogRapyutaCore: [RRROS2GameMode.cpp@31, virtual void ARRROS2GameMode::InitGame(const FString &, const FString &, FString &)] INIT GAME [UEDPIE_0_Turtlebot3_benchmark/Turtlebot3_benchmark] - Options: 

[2025.01.10-16.12.11:632][815]LogRapyutaCore: [RRROS2GameMode.cpp@37, virtual void ARRROS2GameMode::InitGame(const FString &, const FString &, FString &)] NUM OF CPU CORES: [16] - WITH HYPERTHREADS: [32] - RECOMMENDED NUM OF WORKER THREADS: [30]
[2025.01.10-16.12.11:632][815]LogRapyutaCore: Display: [RRROS2GameMode.cpp@39, virtual void ARRROS2GameMode::InitGame(const FString &, const FString &, FString &)] ShouldUseThreadingForPerformance: 1
[2025.01.10-16.12.11:632][815]LogRapyutaCore: Display: [RRBaseRobot.cpp@225, void ARRBaseRobot::CreateROS2Interface()] [BP_TurtlebotWaffle_C_3] IsNetMode: 0
[2025.01.10-16.12.11:632][815]LogRapyutaCore: Warning: [RRGameSingleton.cpp@77, static URRGameSingleton *URRGameSingleton::Get()] RF_Standalone!!
[2025.01.10-16.12.11:632][815]LogROS2Node: [ROS2NodeComponent.cpp@103, void UROS2NodeComponent::Init()] [ROS2NodeComponent_0] start initializing..
[2025.01.10-16.12.11:632][815]LogROS2Node: [ROS2NodeComponent.cpp@108, void UROS2NodeComponent::Init()] [ROS2NodeComponent_0] rclc_node_init_default

/home/ubuntu/UE5.1/Engine/Binaries/Linux/UnrealEditor: symbol lookup error: /opt/ros/humble/lib/librmw_fastrtps_shared_cpp.so: undefined symbol: rmw_time_equal
```

arjuntheprogrammer, 2025-01-10:
I tried to change the branch to humble - but ran into following issue in that case:
https://github.com/rapyuta-robotics/turtlebot3-UE/issues/91

yuokamoto, 2025-02-05:
Sorry for late response.
Did you execute `./run_editor.sh` without `source /opt/ros/humble` ?

james-yoo, 2025-02-05:
Hi @noahl1,
Try rclUE with 'UE5_devel_humble' branch not 'UE5.3'