rapyuta-robotics/turtlebot3-UE #93: UE5.4 Support

yoodk5257, 2025-02-03:
Hi rapyuta-robotics.

I want to use UE5.4 and ros2 humble. Is it possible?

james-yoo, 2025-02-04:
Hi @yoodk5257,
It will work with UE5.4 and ros2 humble.

yoodk5257, 2025-02-05:
Thank you for the answer!

Would it work if I replace `rclUE` with `rclUE (UE5_devel_humble branch)` and `RapyutaSimulationPlugins` with `RapyutaSimulationPlugins (UE5.4 branch)` in `turtlebot3-UE (humble branch)` and then run it?

yoodk5257, 2025-02-05:
I tried with

turtlebot3-UE (humble)
RapyutaSimulationPlugins (UE5.4)
rclUE (UE5-devel-humble)

ubuntu22.04, ros2 humble

but I failed. 

I cloned turtlebot3-UE (humble) and changed Plugins by Gitclone.

yoodk5257, 2025-02-05:
I got warnings when i build

[image]

[image]

and I got errors when i run

[2025.02.05-04.32.59:726][  0]LogAssetRegistry: Error: Package /home/home/turtlebot3-UE/Plugins/RapyutaSimulationPlugins/Content/Textures/T_Default_ORM.uasset has malformed tag
[2025.02.05-04.32.59:726][  0]LogAssetRegistry: Error: Package /home/home/turtlebot3-UE/Plugins/RapyutaSimulationPlugins/Content/Textures/T_Default_N.uasset has malformed tag
[2025.02.05-04.32.59:726][  0]LogAssetRegistry: Error: Package /home/home/turtlebot3-UE/Plugins/RapyutaSimulationPlugins/Content/Textures/T_Default_D.uasset has malformed tag

[40/43] Compile Module.RapyutaSimulationPlugins.8.cpp
In file included from /home/home/turtlebot3-UE/Plugins/RapyutaSimulationPlugins/Intermediate/Build/Linu

yuokamoto, 2025-02-09:
Could you try jammy_UE5.4 branch and update submodules?

yoodk5257, 2025-02-12:
I used **UE-jammy_5.4**. The robot appears in Unreal Engine, but there seems to be an issue with **FastDDS**.

home@home:~/turtlebot3-UE$ ./run_editor.sh true $(pwd) /home/home/UnrealEngine/Engine/Binaries/Linux/UnrealEditor
2025-02-12 15:13:06.287 [XMLPARSER Error] Found incorrect tag 'userTransports' -> Function fillDataNode
2025-02-12 15:13:06.287 [XMLPARSER Error] Error parsing participant profile -> Function parseXMLParticipantProf
2025-02-12 15:13:06.287 [XMLPARSER Error] Error parsing profile's tag participant -> Function parseProfiles
2025-02-12 15:13:06.287 [XMLPARSER Error] Error parsing '/home/home/turtlebot3-UE/fastdds_config.xml' -> Function loadXMLFile

yoodk5257, 2025-02-13:
I tried with that branch. However, the LiDAR red dot is not visible.
Also, when I try running :  ros2 launch nav2_bringup tb3_simulation_launch.py use_simulator:=False map:=/home/home/turtlebot3-UE/Content/Turtlebot3_benchmark.yaml
```
I got an error saying **"ros2 not found."**

If I manually source ROS2 in my `bashrc` file with: source /opt/ros/humble/setup.bash
and then run the same launch command again, it executes, but I got an error log stating that the **odom topic is not being published**.

![Image](https://github.com/user-attachments/assets/d6aea7c0-52c1-4201-8908-4560cb8a54d7)

![Image](https://github.com/user-attachments/assets/6e3316dd-a4b5-4860-9571-b0763078102d)

james-yoo, 2025-02-13:
@yoodk5257 
If you press 'Start' button, the simulation will start and you will see red dots.
Also, you should source ros2 workspace before launch the Unreal Editor. 
For example,
```
source /opt/ros/humble/setup.bash
cd <path/to/turtlebot3-UE>
./run_editor.sh
```
Then open the new terminal and launch tb3 simulation package.
```
source /opt/ros/humble/setup.bash
ros2 launch nav2_bringup tb3_simulation_launch.py use_simulator:=False map:=<path/to/turtlebot3-UE>/Content/Turtlebot3_benchmark.yaml
```

closed 2025-02-17 by yoodk5257