rapyuta-robotics/turtlebot3-UE #104: Issue With Distributed Simulation

RXS161, 2025-02-26:
Hi,

I'm trying to follow the documentation for running a distributed simulation. I seem to be unable to replicate the demo behaviour. I play the demo using the LargeGround map and play as client. I then use the ros package used in the demo to try spawn a robot. The server logs would suggest that a robot is being spawned, however, no robot appears.  When I play the level in standalone mode, the robot spawning does work though . Where could I be going wrong? I am following this: https://rapyutasimulationplugins.readthedocs.io/en/devel/distributed_simulation.html. I have also attached an image of the server output after I send a spawn entity request using the test package used in the documentation video.

Thank you. 

[image]

yuokamoto, 2025-02-27:
Thank you.
We haven't tested distributed sim long time. Let us check distributed sim setting and fixit.
sorry for inconvinence

RXS161, 2025-02-27:
Ah I see no problem. Thank you for your quick response!