rapyuta-robotics/turtlebot3-UE #105: Physics issue for manipulators

james-yoo, 2025-03-10:
Hi @yuokamoto,

I would like setup a simple 'mobile manipulator'(mobile base + manipulator) for navigation test.  
However, when the 'simulate physics' option is enabled for 'Link0' as shown in the following figure, the manipulator passes through the floor mesh(video below).
[image]  
    
https://github.com/user-attachments/assets/797ec644-aa2a-46aa-902a-78c3b2aa91bd  
  
Other physics-based manipulators (e.g. UR10) show the same issue. Is there a workaround?

yuokamoto, 2025-03-16:
I'm investigating. current status
- if I remove the link and joint after link2, it won't go through floor
- If I use normal Physics constraints component in BP, it works as well. so something wrong in RRPhysicsJointComponent.

mean while, by creating phyiscs joints between arm and mobile robot, it may work

james-yoo, 2025-03-21:
@yuokamoto,

Collisions are disabled due to initialization process, this line:
 https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/blob/01c5cc8d26949a20741bebfb7d17d6cd7c5982b2/Source/RapyutaSimulationPlugins/Private/Robots/RRBaseRobot.cpp#L654

It makes collision profiles 'OverlapAll' here: 
https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/blob/01c5cc8d26949a20741bebfb7d17d6cd7c5982b2/Source/RapyutaSimulationPlugins/Private/Robots/RRBaseRobot.cpp#L703

If I changed the value to true, the robot does not fall anymore.

yuokamoto, 2025-03-24:
Thank you.
It is disabled for initialization.
I will add option not to disable collision, not disable collision if joints have only either parents/child or something