rapyuta-robotics/turtlebot3-UE #106: No camera images with jammy_UE5.5

bastianhjaeger, 2025-03-11:
The repo seems like a great deal!!!

Unfortunately I do not get images from the robot. How to enable image publishing?

Sidenote: Also 
```
ros2 launch nav2_bringup tb3_simulation_launch.py use_simulator:=False map:=./Content/Turtlebot3_benchmark.yaml
```
seems to not work properly (see screenshot), as the TF tree is incomplete.
```
[rviz2-3] [INFO] [1741683542.927129435] [rviz]: Message Filter dropping message: frame 'odom' at time 23.370 for reason 'discarding message because the queue is full'
[rviz2-3] [INFO] [1741683542.958812393] [rviz]: Message Filter dropping message: frame 'base_scan' at time 24.250 for reason 'discarding message because the queue is full'
[component_container_isolated-4] [INFO] [1741683542.977551258] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-3] [INFO] [1741683543.022473230] [rviz]: Message Filter dropping message: frame 'base_scan' at time 24.280 for reason 'discarding message because the queue is full'
[rviz2-3] [INFO] [1741683543.086980625] [rviz]: Message Filter dropping message: frame 'base_scan' at time 24.320 for reason 'discarding message because the queue is full'

```

OS: Ubuntu 22.04.
ROS2 humble installed.
Using branch `jammy_UE5.5` and unreal `5.5.3`.

[image]
[image]

yuokamoto, 2025-03-11:
Thanks for you comment.
- Default turtlebot does not have camera. you need to attach camera. ref https://github.com/rapyuta-robotics/rclUE/issues/95#issuecomment-1600582298
- Did you provide initial pose from rviz?

bastianhjaeger, 2025-03-11:
> * Did you provide initial pose from rviz?

No. I ran 
```
roslaunch nav2_bringup tb3_simulation_launch.py use_simulator:=False map:=./Content/Turtlebot3_benchmark.yaml
```
as documented.